libhal-google/libhal #78: frequency * percent => duty_cycle

kammce, 2022-02-02:
Implement an operator overload for multiply such that:

```C++
duty_cycle operator*(frequency, percent)
duty_cycle operator*(percent, frequency)
```

herculanodavi, 2022-04-15:
Hi, I'm new to open source and wanted to start contributing to your repo.
Maybe this issue is a good first one? 
.take

kammce, 2022-04-15:
Go for it. This is a good one to start with 😄

herculanodavi, 2022-04-16:
Turns out I'm not sure of the intended result here. 

I first thought I could reuse the `calculate_duty_cycle(frequency, percent)` logic, but that method has 3 pieces of information (target freq, driving freq and duty cycle ratio).

I imagine that the `frequency` parameter int the `operator*` is the target frequency, so how should I calculate the total number of cycles that duty_cycle should have without having a driving frequency value?

kammce, 2022-04-18:
Yeah thats a good point and honestly something that I've been thinking about as well when you picked up the issue. Because my original intention was that this would be used for PWM signals, where you have a clock driving the PWM peripheral and the frequency you intend the output to operate at. So one could imagine a 1MHz clock driving a timer that controls the output of a pin that has a frequency of 1kHz.

So yeah, this issue may not even make sense. 😅

kammce, 2022-04-18:
Yeah, I think this particular idea came up early on before we understood how `calculate_duty_cycle` was going to work. So this issue is actually obsolete. Thanks for the PR and looking into this. Closing this issue as obsolete and will do the same for the PR.

closed 2022-04-18 by kammce