libhal-google/libhal #137: embed::temperature

kammce, 2022-03-25:


kammce, 2022-03-30:
@MaliaLabor put a comment here

MaliaLabor, 2022-03-30:
comment

kammce, 2022-04-15:
Going to remove @MaliaLabor from this assignment as this particular task requires defining how units will be handled in general across the ecosystem and that seems like a pretty daunting first task. I, myself, do not know how units will act going forward so this is something that I should tackle in a separate issue.

kammce, 2022-04-24:
Was going to simply use nanokelvin for this particular library but it looks like mp-units does not support prefixes for Kelvin. I'm not sure why this is. Going to ask in the issues to see why they choose this and what the utility of no_prefix is in this case. It makes sense disallow milli-foot or mega-inch because a more basic unit would be meters for which there are prefixes for.

kammce, 2022-04-24:
Looks like this is the reason, its just not supported yet. https://github.com/mpusz/units/pull/232

This interface is basically blocked by this, unless a technical-debt like work around is made.

kammce, 2022-08-22:
Refer to for details of what was there before #407

commit 5694d402 (2022-04-22): Add embed::temperature

Resolves #137
commit ea516e5a (2022-04-22): Add embed::temperature

Resolves #137
commit 31885728 (2022-04-22): Add embed::temperature

Resolves #137
commit c39bf29b (2022-04-27): Add embed::temperature

- Small fixup in accelerometer.hpp
- Patch thermodynamic-temperature.h based on:
    - https://github.com/mpusz/units/commit/5d82969e0a2c7cd9faf0de7e4482839c40291494
- Add nano, micro, milli, and kilo kelvin units

Resolves #137
closed 2023-01-21 by kammce