libhal-google/libhal #138: embed::linear_analog_temperature_sensor

kammce, 2022-03-25:
Many analog temperature sensors have a linearly proportional relationship between the voltage and the temperature. A generic temperature sensor can be constructed with an `embed::adc` and some sort of linear proportion mapping structure. The class would look something like this:

```C++
namespace embed 
{
template<linear_proportion_t Proportion>
class linear_analog_temperature_sensor : embed::temperature
{
  linear_analog_temperature_sensor(embed::adc & p_adc, /* anything else needed */);
  /* ... other apis ... */
};
}
```

Examples of ICs with linear output voltages proportional to 

- [TMP35/36/37](https://www.analog.com/media/en/technical-documentation/data-sheets/TMP35_36_37.pdf)
- [LMT85](https://www.ti.com/lit/ds/symlink/lmt85.pdf)
- [MAX6516 - MAX6519](https://www.mouser.com/datasheet/2/256/MAX6516-MAX6519-369646.pdf)
- [TMP23x-Q1](https://www.ti.com/lit/ds/symlink/tmp236-q1.pdf)

MaliaLabor, 2022-03-27:
I'm on it!

kammce, 2022-04-15:
Unassigned @MaliaLabor for the same reason as posted here #137

kammce, 2022-04-17:
Depends on #137