libhal-google/libhal #206: Decide how mocks for serial communication should be handled

kammce, 2022-04-29:
Unlock many other interfaces, serial communication can be quite complex. 

Where interfaces like pwm and dac can simple be spied on and their inputs stored for later inspection. Interfaces like adc and temperature generally require a sequence of return values to be mocked.

Serial communication protocols can have a multitude of real world protocols for how they communicate. SPI and I2C tend to do things via a register based memory map, where the first few bytes in the transaction are the register address and the rest of output bytes are the data to be written or the first few bytes are the register address and the bus is reading data out of the device. This isn't always the case though and some devices are write only and never return anything, these can be easily spied on. Some devices always simply return information, these can be easily mocked. But some have more bi-spoke ways of communicating. 

Uart can be wildly varied in how the port is used to transmit data. Sometimes it its read only. Sometimes it write only. Sometimes its address based. Some times is command based. Sometimes it is a combination of these things. 

CANBUS is based on message IDs, but many times rather than using up multiple message IDs, they simply put the command/address in as the first byte of the, up to, 8 byte CAN data.

So deciding how each of these should be handled is an interesting and non trivial question.

kammce, 2022-04-29:
# For I2C what are the kinds of communication protocols around?

## write-then-read single to 8-bit address space with 8-bit data
- [MPU6050](https://invensense.tdk.com/wp-content/uploads/2015/02/MPU-6000-Datasheet1.pdf) and most other accelerometers

## write-then-read single to N*8-bit address space with N*8-bit data

## write only, no read

-  Display drivers like: [SSD1306](https://cdn-shop.adafruit.com/datasheets/SSD1306.pdf) oled display

## single byte/register write & read
- I2C Multiplexers: [tca9544a](https://www.ti.com/lit/ds/symlink/tca9544a.pdf)

## Bespoke
- TMP102: 
    - to read you must do a write transaction to set address pointer, then a read for the number of bytes needed rather than a write-then-read.
    -  8-bit address space, 16-bit data space

herculanodavi, 2022-05-01:
I think this one is similar to the TMP102: [TLI493D](https://www.infineon.com/dgdl/Infineon-TLI493D-A2B6-DataSheet-v01_30-EN.pdf?fileId=5546d46262b31d2e0163162ca93b65fb)
You don't do a write-then-read exactly, but you first write and start reading without the stop bit.

[image]

kammce, 2022-05-01:
@herculanodavi thanks for the additional info. I think I remember some weird chip having a very similar way of behaving. Its quite unfortunate because this type of transaction is not something that many i2c peripheral blocks support too easily. Its also interesting how 1 transaction gives you all of the data all at once.

kammce, 2022-05-01:
Spent a few days thinking over this and I think I have an idea of what would work well here. Something like this for 8-bit register address and 8-bit data. It's a tad complicated, but I think it gives the flexibility needed to handle most situations.

Writing up a godbolt example to post here.

kammce, 2022-05-01:
The main idea is that we would use something like this:

```C++
using device_address = std::byte;
using register_address = std::byte;
using register_responder = std::function<auto_increment(operation, std::byte *)>;
using memory_map = std::unordered_map<register_address, register_responder>;
using i2c_bus = std::unordered_map<device_address, memory_map>;
```

The idea is that we have a map of i2c devices. The key to each i2c device is its i2c address. Each i2c device is represented by a map of registers, where the key is the register address and the values are callback functions. These functions take an operation and a pointer to a byte and returns an auto-increment enumeration. The operation tells the function if this is a write or read operation and the byte pointer is where to store the information. The auto increment enumeration tells the i2c mock if it should move to the next register or not. In the case of TMP102, register pointers do not move unlike most other systems. 

- For READ ONLY registers, write operations on those registers should throw exceptions as part of their functions.
- For polling a register, a functors or lambda could remember the number of times its been called. This will allow the register to stay a specific value for a number of read operations before finally changing and breaking the poll condition.
- For WRITE ONLY registers, read operations will throw an exception, but write operations are fine. The data in that register can be inspected in the unit test
- In more weird situations, like the TLI493D, the register_responder can manage how exactly it handles written and read data.

kammce, 2022-05-01:
I believe that such an approach works above works well for i2c. I think it should work well for SPI, but I'm not certain. 

Serial may even be more varied to the point where it simply makes sense for each unit test for a device to simply write up their own mock for each situation. For example with the ESP8266, a unit test will simply need a mock that knows how to respond to each command. In a way, this mock is a dumbed down simulation of of the esp8266 hardware written in software.

For now I think that this is a decent enough approach to get started with the i2c and spi mock. @herculanodavi how do you feel about this?

kammce, 2022-05-06:
[image]

Having each register be implemented by a callable allows a great amount of flexibility such as:
- Simply saving a value written to it and returning that value
- Polling behavior based on subsequent reads of the register
- Throw exceptions if a register is accessed in a way that is invalid for the actual hardware, like writing to a read-only register or vise versa
- Sets of registers can have coupled behavior using lambdas or functors. 
    - As an example, lets say we are mocking/simulating an accelerometer, we can have the X, Y and Z registers return 0 when the enable bit within a control register is 0 and have it return a predetermined set of values when that bit is set to 1

The idea would be to make a set of helper objects that handle this complex behavior and the user just has to pass some parameters to those object as well as the embed::mock::i2c object.

kammce, 2022-05-08:
Had a discussion with @herculanodavi today at noon PST about how mocks should be handled between i2c and spi. The conclusion we came was that a mock for I2C, SPI and any serial protocol is too varied make a general purpose mock make any sense. Expectation based mocking can be very painful to work with as it tends to result in tests that are tightly coupled to exactly how the driver code is written rather than the expected behavior of the driver. 

The solution that we decided on is that device simulations should act as the mock/fake for testing. Having the test software pit itself against the simulated driver should result in testing code behavior and outcomes.

An example may be something like this:

```C++
namespace embed::fake {
class mpu6050 : public embed::i2c {
  // implements transaction in such a way that it emulates the register file within the device
};
}

"mpu6050 construction"_test = []() {
  // Setup
  embed::fake::mpu6050 fake(/* ... */);

  // Exercise
  embed::mpu6050 mpu(fake, /* ... */);
  
  // Verify
  expect(fake.is_initialized());
};

"mpu6050 read()"_test = []() {
  // Setup
  embed::fake::mpu6050 fake(/* ... */);
  embed::mpu6050 mpu(fake, /* ... */);
  std::array<embed::accelerometer::sample, 3> samples;

  // Exercise
  mpu.read(samples);
  
  // Verify
  expect(fake.is_initialized());
  // Verify: the fake generate accelerometer values and records what registers have been read out
  // This step confirms that the samples returned equal the samples passed in. 
  expect(fake.verify(samples));
};

// etc
```

And because we believe this to be a more scalable approach for each driver we no longer need generic mocks for i2c, spi, serial and can. We may make helpers that make writing the simulations easier, but we will write those as we reach the need for them.

Closing this issue:

Resolves #5 
Resolves #211 
Resolves #6

closed 2022-05-08 by kammce