libhal-google/libhal #263: Add embed::poll(timeout, polling_function)

kammce, 2022-06-20:
Provides a utility that polls a function until it either indicates that it has completed its work or , then the function throws `embed::error::timeout`, otherwise it returns success.

SameerAzer, 2022-06-28:
I can help with this one

SameerAzer, 2022-07-21:
Does this belong in libembeddedhal/timeout.hpp ?

kammce, 2022-07-21:
Yes, it should be in `timeout.hpp`

SameerAzer, 2022-07-22:
What's polling_function's signature? boost::leaf::result<bool>(void) ??

kammce, 2022-07-22:
`boost::leaf::result<void>` type can be used to indicate success or if an error was thrown. `embed::poll` executes a polling_function until it is either finished, successful poll, or until the timeout times out, return `boost::leaf::new_error(embed::error::timeout{})`.

I'm not sure when poll would return true/false otherwise.

kammce, 2022-07-22:
Oh sorry, I misread this. The polling function should return boost::leaf::result<bool> where the success value of false means that the polling function has not yet finished its work, true means its work is finished and an error will be transmitted upwards.