libhal-google/libhal #293: Alpha Launch Interface Review

kammce, 2022-07-16:
Go through each and every alpha launch interface and consider the following:

1. Give rationale for why each API must be there, no API is given an exception, and APIs that do not fit should be eliminated.
2. Give rationale for why an API MUST have a `boost::leaf::result<T>` return type and determine if any APIs can get away without them
3. Give rationale for why each parameter must be in the API

The list of interfaces to go through are the following:

- [ ] adc
- [ ] can
- [ ] counter
- [ ] dac
- [ ] i2c
- [ ] input_pin
- [ ] interrupt_pin
- [ ] motor
- [ ] output_pin
- [ ] pwm
- [ ] serial
- [ ] servo
- [ ] spi
- [ ] timer

kammce, 2022-07-24:
From looking at these interface for so long, I have but a few that I feel questionable about. I'll go over each of these particular interfaces and describe how I'm feeling.

# embed::i2c::configure()

The problem with configure is that it contains clock rate. Making this particular API public means that any arbitrary driver can change the frequency of the I2C peripheral. This is a problem because we can have a situation where devices can support different clock rates from each other. 

For example, say we have 3 drivers, two of which support 100kHz and one that supports 400kHz. Each driver set the i2c configure to their preferred high speed clock rate. Let's assume that in this application the developer did not specify a consistent clock rate for each driver even though they support specifying one over the default. So each driver will set the clock to their maximum rate. Running a bus faster then some devices can handle is against the protocol rules for I2C and the behavior is undefined. 

This begs the question, should this API be apart of the interface or not? The current purpose of this API is to be used with `embed::bus<i2c>` where each driver can get a profile i2c driver to which they can set to their needs. Anytime those profile i2c drivers are accessed, the i2c bus is reconfigured to their specified clock rate. Unlike the version of bus that works for `embed::spi` the i2c variant would NEED an i2c bus multiplexer to keep the effective bus clock domains separate. 

So the choice is to keep this as an API for configuring the device or make setting the clock rate driver specific with their own rules and behaviors. `embed::bus` for i2c could still work but at a higher cost by using `std::function</*...*/>` 

- Keep current API
- Make configuration device driver specific
    - `embed::bus<i2c>` can still work but at a higher cost and using polymorphic functions `std::function</*...*/>`

# embed::serial::attach_interrupt(/* ... */)?

We currently do not have this API but should we make it a hard requirement? The idea of attach interrupt would be that when the capacity of the DMA or interrupt driven serial port fills to a particular percentage, a user defined callback is executed. That callback can then be used to handle pulling data out of the buffer. 

When I think about this problem I think about the esp8266 driver. The main thing to understand about that driver, without actually looking at the specifications is that, when you execute a request to a server the amount of data you get back can be enormous. The esp8266 could inject its own handler that manages pulling data out of the serial port and into its own buffer. 

BUT! Not all DMA based MCUs can support interrupting at arbitrary fill percentages so thats another thing to be concerned about. What other hand I drop based systems for cereal and immediately tell you when they hit a certain limit.

# embed::input_pin & embed::output_pin

Should there exist an interface just for supporting a pin that can have its direction switched from input to output.

This particular functionality would be useful for developing a bit-bang i2c driver. 

In particular, output_pin has a interesting capability from its current API. If you set it as open collector and set the active level to `true` (lets assume this means high voltage), and then use the `level()` to get back the state of the pin, this is effectively an input pin. So we already seem to have a means of creating an input/output pin. But not sure if it makes sense to have its own type.

kammce, 2022-07-25:
# embed::i2c::configure()

From my experience working with Linux systems and consumer products, being able to have multiple devices work at differing speeds with i2c may be critical. 

Considering that this would only ever work with a I2C multiplexer, maybe it stands to reason that there should be some work put in when designing the bus class to incorporate I2C multiplexers in that design directly. It may even make sense to have a I2C multiplexer Interface with a single function allows you to change which bus you're currently using. That also may be too extreme. It could also be a polymorphic function that takes an index and switches the bus for you. Something like:

```C++
auto i2c_driver = /*...*/;
auto i2c_mux(i2c_driver);
auto i2c_bus = embed::bus<embed::i2c>(i2c_driver, 
[&i2c_mux](int p_port) -> boost::leaf::result<void> {
  return i2c_mux.port(p_port);
});
```

herculanodavi, 2022-07-25:
# embed::input_pin & embed::output_pin
Should we then not have a `push_pull_pin` and `open_drain_pin`? Naming is very much improvable, but the behaviors are so different that I would say there should be two different interfaces. Then the `open_drain_pin` could also have a `level()` function.

kammce, 2022-07-25:
> # embed::input_pin & embed::output_pin
> Should we then not have a `push_pull_pin` and `open_drain_pin`? Naming is very much improvable, but the behaviors are so different that I would say there should be two different interfaces. Then the `open_drain_pin` could also have a `level()` function.

We could do this, but there is one capability I forgot to mention about the output pin. To turn it into an input, you set it to open drain + you set its level to true. But when you want to set it back to push pull, as a means to transmit information, you can turn off open drain and have full control over the bus. Not 100% sure if there is a real use case for this. This particular behavior is found in the I3C specifications and is used to make transmitting data faster.

But to your point, would there be just those two interfaces or would there be `input_pin` (assumed push pull), `output_pin` (also assumes push pull), and `open_drain_pin`? Or something different?

herculanodavi, 2022-07-25:
wrt I3C I have no experience, but I mean exactly that as a proposal

kammce, 2022-07-25:
I really like the idea of having a soft driver implement input-output pin functionality. 

But:

> behavior (push-pull -> open drain), this would likely trigger some `configure` call, thus we might be looking at a software driver similar to the `embed::input_output_pin`.

Would this mean that the output_pin API would still have a means to configure the pin's open drain state? If so, is `open_drain_pin` a more constrained interface? Or could it also just be a soft driver that limits the APIs for the pin? Because if we have an `open_drain_pin` it would make sense to eliminate the `open_drain` configuration for `output_pin` to reduce overlap.

kammce, 2022-07-25:
Hmmm, may also make sense to name `input_output_pin` as `tristate_pin` although thats more cryptic 🤔

closed 2022-09-08 by kammce