libhal-google/libhal #328: Add `embed::minimum_speed_i2c`

kammce, 2022-07-25:
# API

Implements `embed::i2c` interface.

```C++
class minimum_speed_i2c : public embed::i2c {};
```

# Problem

Consider the following:

```C++
auto& i2c = /* ... */;
auto device1 = device1::create(i2c); // 200kHz speed
auto device2 = device2::create(i2c); // 100kHz speed
auto device3 = device3::create(i2c); // 400kHz speed
```

Each device sets the i2c clock rate to their highest speed which differ from each other. An I2C bus must only be operated at the speed for which the slowest device can support. The last factory function sets the speed last and that could be a value greater than what other devices can support. This can result in errors with the lower speed devices. 

# How this solves it

Rather than passing the i2c driver directly to the devices pass them an `embed::minimum_speed_i2c` which ensures that the clock rate of the i2c driver is always set to the lowest speed of any device driver that calls `::configure()`.

Example:

```C++
auto& i2c = /* ... */;
auto minimum_speed_i2c = embed::minimum_speed_i2c::create(i2c); 
// Configures to 200kHz speed
// minimum_speed_i2c configures i2c to 200kHz. 
auto device1 = device1::create(minimum_speed_i2c); 
// Configures to 100kHz speed
// minimum_speed_i2c configures i2c to 100kHz. 
auto device2 = device2::create(minimum_speed_i2c);
// Configures to 400kHz speed
// minimum_speed_i2c will ignore this call since it would be greater than the current 
// clock rate.
auto device3 = device3::create(minimum_speed_i2c); // 400kHz speed
```

kammce, 2022-07-25:
Can you take this @liznguyen12 for iteration 9?

liznguyen12, 2022-07-26:
.take

PR #370 (2022-08-09): Add hal::minimum_speed_i2c
Resolves #328
commit 59905d4d (2022-08-09): Add hal::minimum_speed_i2c

Resolves #328
closed 2022-08-09 by kammce