libhal-google/libhal #548: Determining how many bytes were consumed by a stream

kammce, 2022-11-18:
Streams return the amount of data remaining, but do not give back information about how many bytes were consumed by the pipeline. This information is helpful as it can be used to shovel the data that was consumed into other parallel pipelines.

Currently:

```C++
auto remaining = input | fill | find | parse<unsigned>;
// We get how much is left of input, but we don't know how much of input was consumed by the pipeline
```

MaliaLabor, 2022-11-19:
.take

kammce, 2022-11-19:
One thing to look into is the `std::distance` function.

But we should also do a check to ensure that the two spans that are given, one of them is a subspans of each other.

kammce, 2022-11-20:
Ooo, maybe a wrapper around the pipeline like so:

```
auto info = proper_wrapper_name(input | fill | find | parse<size_t>);
```

Info could be something like:

```c++
struct S {
  span remaining;
  size_t bytes_read;
  // Other useful information about the pipeline
}
```