libhal-google/libhal #632: Reconsider `hal::servo::position()`

kammce, 2023-01-20:
Currently it uses a value from -1.0 to +1.0, but this is so abstract that it may not even be understandable for users or driver developers. A better alternative would be to simply use angle in `hal::degrees`.

starphys, 2023-01-30:
.take

kammce, 2023-03-07:
Discussion on discord:

> Khalil Estell — 02/19/2023 4:53 PM
Evan Astle (they/them) so I've been thinking.
[4:54 PM]
What if we consider documentation into the fold. Like, we make an application and we state in the usage guidelines, "servo must have at least range from X to Y" and "zero position should be connected to this joint as so (shows image)".
[4:56 PM]
I still don't know if going past the possible angle values ought to generate an error or simply silently clamp.

> Evan Astle (they/them) — 02/19/2023 4:56 PM
I like that idea
[4:57 PM]
My gut reaction is that an error is better because it gives software the choice of handling it or ignoring it
[4:59 PM]
But obviously clamping has the benefit of protecting the hardware
February 20, 2023

> Khalil Estell — 02/20/2023 11:38 AM
True, but if we are having implementations throw errors then we can instruct them in the documentation to also not move the position of the servo shaft if they are commanded outside of their range.

> Evan Astle (they/them) — 02/20/2023 1:52 PM
Yeah that's also my thinking, errors give more flexibility and let implementations make their own decisions about how to handle input outside of the supported range

> Khalil Estell — 02/20/2023 1:59 PM
Well I don't know if we want to allow implementations to decide on how to handle inputs outside their range. We need to define clearly, without ambiguity how exactly the system should act when all parties are acting correctly, without error, how to handle inputs outside of their range. By having such a guarantee, it allows developers to better know how to handle these sorts of errors or events
[1:59 PM]
Think of this like making laws but I'm software
[1:59 PM]
In*

> Evan Astle (they/them) — 02/20/2023 2:12 PM
Gotcha, that makes sense
[2:13 PM]
Even with that I think errors have advantages for control software

> Khalil Estell — 02/20/2023 2:36 PM
Oh absolutely. We define the expected behavior and then give feedback about the issue to the program.

kammce, 2023-03-07:
I think the solution here is to do the following:

- Make the API for servo position -> `hal::servo::position(hal::degrees)` such that the application can pass the desired angle and the servo must obey this request.
- Add a soft driver called `hal::offset_servo` which implements servo but offsets the range of the servo. For example, if you have a servo that can move from 0 to 90 degrees but an application want to work with a servo from -45 deg to 45 deg, because the range is actually the same just that the 0 position isn't in the right location, the offset_servo can manage remapping inputs from one servo to the next servo. 
- Add to documentation of the servo API that an angle passed that goes outside of the available range performs no movement of the servo rotor and throws an error that returns `std::errc::invalid_argument` and a `hal::servo::range_error` which includes the min and max of the servo. This error can be teased out by supplying +INF or -INF.
- Put up application notes discussing that applications that use servos must make it clear what is their mininum range in terms of degrees and what their min and max degrees they are working with. For example, we have a rotating plate that needs to move from 0deg to 270deg. This way, a developer of the application can feed an `hal::offset_servo` with the correct mapping.

closed 2023-03-07 by kammce